gigibu5/OctoLight #1: suggestion:change indicator to show actual value

Hairybiker, 2021-01-20:
Be nice if the light was off when the printer led was. & vice versa

gigibu5, 2021-01-20:
Tought about that, might even implement it. But then again, you use the light to see the printer on camera, so it's kinda redundant :slightly_smiling_face:

Hairybiker, 2021-01-20:
I don't use a camera...
But it was purely cosmetic anyway

jumon-zz, 2021-01-20:
I know you just published and wanted to say this is a great idea for a future enhancement.  It would be great if it showed status (on/off) with the icon. Also, being able to change the icon to another one perhaps (like a fan).  I think many people might use this for other items so a light, fan, and generic power plug might be great icons.

gigibu5, 2021-01-20:
Great idea, could probably implement, but probably with Unicode emojis. Do you know any got emoji font's, maybe I can get some other icons from there?

cp2004, 2021-01-20:
OctoPrint uses FontAwesome 5 (since OctoPrint 1.5.0, FA 4 before then), see if any of those icons fit your use case.

gigibu5, 2021-01-20:
Great, that's very helpful.

adamboutcher, 2021-02-05:
fas fa-lightbulb
vs
far fa-lightbulb

might be simple enough to implement and show the difference...

StaMau, 2021-02-20:
It will be for sure a cosmetic improvement but also will be usefull, because sometime, from the camera si not possible to understand if the light is actually swithed on. For example during the day I'm not able to understand if the light is switched on or not, because of daylight. Thank you for your job.