gigibu5/OctoLight #2: Plugin does not show up

MDSign75, 2021-01-20:
Hello,

I've just downloaded and installed your plugin to control my raspi-case fan which is connected to Pin1 and 9 of your GPIO Diagram. 
Unfortunately the plugin does not show up - neither in my settings page nor in the top bar (lightbulb)

- Is there anything i have to do to get the plugin running?
- Is it basically possible to switch these Pins (3,3V & GND) to interrupt the powersupply to my fan?

Would appreciate your help.

Thanks in advance
Maik

gigibu5, 2021-01-20:
Hey, the user who merged a plugin wrote this yesterday on the Octoprint Discord:
> Yes, it seems to be showing properly now. One weird thing I'm not sure on is that after install it's triggering a reload, but not a restart of OctoPrint. @cp2004 I thought TemplatePlugin mixin automatically would trigger that, thoughts? This won't prevent me from merging your submission, just you may end up getting some support requests relative to it's not showing up in the navbar or settings right after install.

It seems there's a bug in the web interface. Have you tried restarting the Octoprint service? You can do that through the UI or typing this into your Raspberry Pi terminal: `sudo systemctl restart octoprint`

The answer for the other question, this plugin only controls the GPIO pins, in the settings, you can set which GPIO it controls. As I can see, your fan is connected to pins 1 and 9 (3.3V and GND). 3.3V pin is not controllable, its a constant source of power. On the Raspberry Pi the pins marked in orange (GPIO XX) are controllable, but my plugin uses the board naming scheme (the physical numbering) of the pins. For example, if you want to control something in `GPIO 27`, that means you must set the setting to pin 13. However, I do not recomend running a fan directly from the GPIO, it may damage the RPI. I suggest controlling a transistor or something similar for turning on and off the Fan. For more information on how to do that, just ask, but it will require external circuitry.

cp2004, 2021-01-20:
If you add `RestartNeedingPlugin` to the mixins that you implement (where `TemplatePlugin` etc. are), that will force a restart when the plugin is installed. Otherwise I will check, there is probably some caching somewhere to make the UI faster.

gigibu5, 2021-01-20:
Will try that

MDSign75, 2021-01-20:
Hi gigibu5,

thanks for your quick and helpful reply, I haven't restarted my Pi since the installation of OctoLight - I've "trusted" the restart following the installation process. Maybe I'll do this now :)

Another "Thanks" for clarifying the GPIO question and your warnings about damaging my Pi. Is there any safe way to get stable 3,3V out of the Pi board? I made a 3D printed fan-lid for my Pi case and for now it seems to work fine with power from the mentioned pins. I've been thinking about controlling the fan by an extenal relais tied to the GPIOs, but that would need an extra power supply and would not be that "sleek and compact"... The fan is designed for 5V and I'm running it at 3,3 volts because of the generated noise ;) That's also the reason for my wish to control the state of the fan - at night the fan is still a bit noisy... Problems you have, when your gear stands in your bedroom :D

Maybe you can give me some advice for the internal powersource - any help is appreciated :)

Thanks again
Maik

gigibu5, 2021-01-20:
Hi MDSign75,
The problem with controlling the fan from GPIO is the high current needed by the fan, the output voltage is 3.3V. Micro-controller pins do not provide much current, that's why you would need some external circuitry. The simplest solution would be, to connect a fan through a relay module. [image]. If you need any help with the wiring, let me know.

MDSign75, 2021-01-20:
Hi gigibu5,

thanks for your advice! Do you have any information about what amount of current in mA "could" be safe over 3,3V. If this value would be too low for may or any other fan, i surely would switch to an relay module with external power like you mentioned. I think I do have some kind of these (f)lying around from my arduino Uno R3 starter kit... ;)

gigibu5, 2021-01-20:
I do not know the specs from the top of my head, but everything under 20mA should be fine. It's definitely not enough to drive a fan. That relay module from the kit would be perfect.

MDSign75, 2021-01-20:
Okay, cool - then I'll check the consumption of the fan and use one of these relay modules, or switch to an external powersupply (maybe 5V from USB) if the fan needs way more than this 20mA...

Many thanks for your patience and good advice!
Maik

gigibu5, 2021-01-20:
5V from the USB port is the same as 5V from the Raspberry Pies GPIO. No need to use an USB plug.

MDSign75, 2021-01-20:
Okay, so it would be okay, if I use Pin2 (5V PWR) instead of Pin1 (3.3V PWR) to get the full amount of mA that comes via USB or is the circuit "through" the Pi (5V to any of the GND) the limiting factor?

gigibu5, 2021-01-20:
Yes, if you use the 5V pin you would get the full current from the PSU, bit you would still need a relay to controll that from the GPIO pin.

MDSign75, 2021-01-20:
Yep, I think I got it :) 5V to GND via GPIO pins to the relay and one GPIO pin to trigger the relay, right?

gigibu5, 2021-01-20:
I drew a diagram for you, I hope it helps:
[image]

OxygenGS, 2021-01-20:
Ah that is amazing! ty that helps me visualise :D

gigibu5, 2021-01-20:
No problem :slightly_smiling_face:

MDSign75, 2021-01-20:
Wow, cool - that does help a lot :) Thanks again - never thought i would get so much support!

closed 2021-01-22 by gigibu5