gigibu5/OctoLight #3: The button in the navigation does not behave as a button

mirco71, 2021-01-20:
Hi,
I installed the plugin and restarted the raspberry afterwards. The button is present in the navigation bar and a port can be configured. Unfortunately the button cannot be pressed. The mouse does not change to as it does for the other buttons of the navigation bar. 
In addition I connected an LED to the configured pin (not using GPIO naming) but the LED does not light even if I change to inverted. 
Can you please give a hint.
Thanks

gigibu5, 2021-01-20:
Yes, I am aware that the apears not to be clickable. I have to add some CSS lines to make convert the select text cursor into a click button cursor. Anyway, the button is clickable. 
Did you check if the LED is correctly orianted? For default configuration: PIN 13 -> 330 ish ohm resistor -> LED anode (long leg), other leg -> GND

mirco71, 2021-01-20:
Thanks for the quick response! Sorry, it was my fault! I did use the wrong port on the raspi. I do not know why I did not realize it before. LED can be switched on an off as expected!!!

closed 2021-01-20 by mirco71