gigibu5/OctoLight #4: What leds can a raspberry pi 4 handle?

OxygenGS, 2021-01-20:
Hello,

I saw this addon popping up in the news on Octoprint.
I was almost printing a light bar already for my Ender 5 Pro.
I did look at the page here and the picture/scheme.

Do the led's get power from the PI? or are they getting power from somewhere else?
I really want to use it, its an awesome plugin!

gigibu5, 2021-01-20:
The plugin enables controlling GPIO through the Web interface. If you use a single LED, it's fine if you power it from the GPIO, but for more LEDs, you should probably think about some external circuitry, for example a transistor as a switch or a relay module. More info in this issue https://github.com/gigibu5/OctoLight/issues/2#issuecomment-763652674

OxygenGS, 2021-01-20:
Ok. So i need to connect 1 cable from the leds to the IO pin, and can use for example a phone charger for the 5v and ground? Will it work (turn on and off)?

gigibu5, 2021-01-20:
You need to connect the anode (long leg) of an LED with an resistor to GPIO pin and the other side of the LED to GND. Here is a picture from the internet: [image]

OxygenGS, 2021-01-20:
This is complicated. So i cannot just use a phone charger that is 5v and connect 1 wire to the io pin to control it and the plus and minus to the power supply  (phone charger)?

gigibu5, 2021-01-20:
Sadly, that's not how electricity works :slightly_smiling_face:. If you can explain what you are trying to do (what light source) it would be apriciated and I can get you a list of parts.

OxygenGS, 2021-01-20:
Well i would like to get a LED strip of about 30 CM atached. And i would LOVE it when i could control it with octoprint with your plugin. the 30CM led strip will be normal white light, so no RGB.

gigibu5, 2021-01-20:
Ok, is your LED strip 5V or 12V. If it's 12V, can you get 12V from your 3D printers PSU?

OxygenGS, 2021-01-20:
My 3d printer has 24v if i am correct. or does it also have a 12v pin?
And then 1 wire from the LED to the PI?

gigibu5, 2021-01-20:
Could you send me your 3D printer model and which LED strip you intend to use?

OxygenGS, 2021-01-20:
I have a Ender 5 Pro. The LED strip i wanna use i am not sure about yet. Because i dunnow what is good and if i need 5v or 12v.
But i wanna use a led strip of 30CM

gigibu5, 2021-01-20:
Your best option is to get a white 24V white LED strip and a 3V relay module. If you search for the parts on eBay, Aliexpress or your local Electronics shop and send links, I can confirm if they would work.

OxygenGS, 2021-01-20:
Ok nice. so i will get a 24v led strip. And this will do? https://prnt.sc/xb2rsc

gigibu5, 2021-01-20:
Yes, this is perfect

OxygenGS, 2021-01-20:
Ok so. i will connect the 24v led strip to the PSU of the printer. And the thirth cable goes from the IO pi of the pi to the Relay?

gigibu5, 2021-01-20:
Your wireing will look similar to this, the only change is that instead of a Fan you use an LED strip and the connection from COM on the relay is connected to 24V on the printers PSU. You also need to connect the printer PSU negative terminal to GND on the Raspberry Pi.

https://github.com/gigibu5/OctoLight/issues/2#issuecomment-763847790

closed 2021-01-22 by gigibu5