gigibu5/OctoLight #6: inoperative plugin

dhawker, 2021-01-21:
Just installed your new plugin with the plugin manager, no problem.
re-booted Octoprint
lightbulb button appears as expected
cursor changes from selection hand icon to indicator icon and cannot select or press button.

suggestion, it would also be useful if the light could be switched on when Octolapse triggers to take picture and off after picture taken.  No need to leave light on continuously when printing overnight.

gigibu5, 2021-01-21:
Hmmm, I don't see how that would be usefull. You want to use the light as a flash?

dhawkerg1, 2021-01-21:
something similar, but the non-operation is a bit more pressing

gigibu5, 2021-01-21:
If I understand correctly, you want to use this with the Octoprints timelapse feature, so you can film timelapses at night?

Member1773, 2021-01-21:
I think he means, that there is no visual feedback when the button (which is an anchor-element) is pressed.

gigibu5, 2021-01-21:
Ah OK, then this is a duplicate of issue #1

closed 2021-01-22 by gigibu5

dhawker, 2021-01-22:
Not the Octoprint timelapse, the Octolapse time lapse which moves the printhead out of shot for the picture. This would be more practical

 

This email has been checked for viruses by AVG.
https://www.avg.com

gigibu5, 2021-01-22:
I probably wont add it, maybe at a later date (v0.2.0 maybe)