gigibu5/OctoLight #12: OctoPrint's system menu

Thierry-Myron, 2021-01-23:
For people using TouchUI or Mobile app, it would be great feature if the toggle was visible in the OctoPrint's system menu (config.yaml)
Cheers

gigibu5, 2021-01-23:
Do you have any documentation or examples on how to do that?

Thierry-Myron, 2021-01-25:
Hi, I have found Octolight Icon on ThouchUI, very small but working.
[image]

For the config on System Menu (Config.yaml) here is what I use:   
    -   action: lighton
        command: gpio -g write 17 1
        confirm: false
        name: Light ON
    -   action: light0ff
        command: gpio -g write 17 0
        confirm: false
        name: Light OFF

Would be great if you could integrate your API call in state of the gpio cmd.

gigibu5, 2021-01-25:
You can make a pull request, but I do not have the time to intergrate it.

closed 2021-01-25 by gigibu5