gigibu5/OctoLight #13: Remote activation and/or automated activation

mariowitdoek, 2021-01-23:
**Is your feature request related to a problem? Please describe.**
Your plugin is just what I was developing (rather still in thinking phase): the pi that activates a GPIO for triggering a relay that put a 12V LED strip above the 3D printer ON/OFF.

**Describe the solution you'd like**
A possibility to switch the GPIO on/off from remote like a REST call or MQTT subscription. Then I can put the light on using home-assistant or NodeRed. Or via those utilities using the (tasmotised) light switch in my wall. 

**Describe alternatives you've considered**
Or just automatically when the printer is printing, but then again when you have a large print ... then the light is always on. 

**Additional context**
Add any other context or screenshots about the feature request here.

gigibu5, 2021-01-23:
You can turn on and off the light by just making a GET request to `hostname/api/plugins/octolight`. It will trigger the code.

closed 2021-01-29 by gigibu5

eloo, 2021-09-12:
For the record: 
its 
hostname/api/plugin/octolight
:+1: