gigibu5/OctoLight #16: Need some help with this relay

OxygenGS, 2021-01-28:
[image]
I did buy this relay. How to connect this to my PSU of the printer and to the pin of the pi for switching it?

derkaputte, 2021-01-28:
Hello, my understanding is that you "only" need to take 3 pins.

VCC - 5VDC (Pi)
GND - GND (Pi)
IN - Pin 13 (Pi, or which one you took)

SGND - is bridged to GND with the jumper, you can choose a seperate way to power up the 5VDC for the relay, but if you take the 5VDC from the Pi, you can use it without SGND.

Best regards and happy printing! ;)

gigibu5, 2021-01-28:
That seems correct, but I think you need to connect VCC to 3.3V on the Pi

OxygenGS, 2021-01-28:
Hmm ok. But i did order a 24v led stip. And it will be powerd by my meanwell PSU of the porinter. The PI gets power from its own source. Maybe this sounds dumb but.. can someone draw on my picture so i can see how to wire it??

derkaputte, 2021-01-28:
[image]

For more details, see " https://robotdyn.com/relay-module-1-relay-5v.html ".

BR Thomas

OxygenGS, 2021-01-28:
Ty so so much :D

gigibu5, 2021-01-29:
Can I close this issue?

closed 2021-01-30 by gigibu5