gigibu5/OctoLight #18: Having two indipentend buttons for two light sources

kjoussen, 2021-01-30:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

kjoussen, 2021-01-30:
I have two LED-Stripes and want to have them switched indipendently.
So two GPIOs and two relays but also two buttons.
Possible?

gigibu5, 2021-01-30:
Duplicate of #11

closed 2021-01-30 by gigibu5