gigibu5/OctoLight #19: Automaticly turn off Octolight when print is done

WorlockM, 2021-02-07:
I would like to have an option in Octolight to turn off my lights when de print job is done.

I kind of achieved this by powering my LED's from the printers power supply, and turnng my printer off with the Tasmota plugin, but the disadvantage of that is that the LED's will turn on again when the printer is on.

gigibu5, 2021-02-11:
Maybe at a later date, curently I do not have a lot of time on my hands.

Syl20-94, 2021-03-20:
If in the same time, you can turn on when starting print seq, it will be cherry on the cake :)

princeje, 2021-12-16:
A work around is to use [Gcode System commands plugin](https://plugins.octoprint.org/plugins/gcodesystemcommands/)  and set up a gcode command to run on print finish that in-turn runs a script to set the GPIO to low.