gigibu5/OctoLight #28: Won't install.

Hairyloon, 2021-04-13:
**Describe the bug**
Seems to install with no errors. Doesn't show up on Octoprint.

**Your software and hardware (please complete the following information):**

OctoPrint 1.5.3 
Python 3.7.3 
OctoPi 0.17.0

 - Browser [e.g. chrome, safari]
Firefox

Razzorback225, 2021-11-08:
Same issue with 

Octoprint 1.7.2
Python 3.7.3
OctoPi 0.18.0

Browser : Brave

ProtomakerSprint, 2022-04-27:
I have the same Won't install issue:

Plugin said:  Successfully installed OctoLight-0.1.3

OctoPrint Version 1.7.3

Pyton 3.7.3

Installed as shown above 

Plugin Manager and direct install URL, (which uninstalled then reinstalled this Plugin) 

Turn off printer after install wait about 2 minutes and turn it back on 

There is no sign of this plugin within OctoPrint 

browser.user_agent: Mozilla/5.0 (Windows NT 10.0; WOW64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/100.0.4896.127 Safari/537.36
connectivity.connection_check: 1.1.1.1:53
connectivity.connection_ok: True
connectivity.enabled: True
connectivity.online: True
connectivity.resolution_check: octoprint.org
connectivity.resolution_ok: True
env.hardware.cores: 4
env.hardware.freq: 1200.0
env.hardware.ram: 914006016
env.os.bits: 32
env.os.id: linux
env.os.platform: linux
env.python.pip: 20.3.3
env.python.version: 3.7.3
env.python.virtualenv: True
octoprint.safe_mode: False
octoprint.version: 1.7.3
systeminfo.generator: zipapi

A Log says 2022-04-27 09:48:54,486 - octoprint.plugin.core - ERROR - Error loading plugin octolight

2022-04-27 10:08:18,586 - octoprint.plugins.pluginmanager - WARNING - The plugin was installed successfully, but couldn't be found afterwards to initialize properly during runtime. Please restart OctoPrint.  < Done this but still no sings of OctoLight Plugin

gigibu5, 2022-04-27:
Did you restart OctoLight after you installed it?

Kiron21, 2022-12-04:
No way to install this plugin.

closed 2023-09-13 by gigibu5