gigibu5/OctoLight #41: Plugin Abandoned?

jneilliii, 2023-08-20:
According to https://github.com/OctoPrint/plugins.octoprint.org/issues/1225, this plugin is abandoned. Posting a new issue request to validate.

thomst08, 2023-08-20:
Hey there,
Just putting in my two cents, I have been using this plugin for a while and tweaked it a bit in a fork.
If the original developer does not wish to maintain this plugin, I would be happy to maintain it, of course with respect to @gigibu5.

gigibu5, 2023-09-13:
@thomst08 I would be happy if you would maintain the plugin going forward. Unfortunately, I have a lot of work to do and no longer have a 3D printer (thanks to a flood at my house), so I don't have a use for this plugin anymore.

thomst08, 2023-09-14:
Hey @gigibu5, thanks for acknowledging, that sounds a bit rough, hope all is well.
@jneilliii, is there some steps to follow to update the links on the Octoprint plugin page?

jneilliii, 2023-09-14:
Yes, [here's ](https://github.com/OctoPrint/plugins.octoprint.org/pull/651) an example PR that demonstrates a completed similar adoption of another plugin. Basically on your fork you'll need to bump version in setup.py and create a new matching release for that version.

thomst08, 2023-09-14:
Excellent, thank you.